sifive/freedom-e-sdk #1: how can i generate bit file ?

rbrlibero, 2016-07-27:


mwachs5, 2016-07-27:
You can download the bitstream (BIT or MCS format) as well as in-depth instructions from https://dev.sifive.com/develop/freedom-e300-arty-dev-kit/

rbrlibero, 2016-07-28:
i don't have this board .
don't you provide fpga verilog source ?
bit files are  built from 
https://github.com/lowRISC/lowrisc-chip
rocketchip or do you have a custom risc-v ?

mwachs5, 2016-07-28:
This SDK repository is targeted for software development on a Freedom E platform. It is currently supported on two hardware platforms: the Diligent Arty Board and Microsemi Arrow SF2+ Development Kit. 

The FPGA images are based on the open source rocket-chip repository, but we do not currently provide the verilog source. You can see that repository's README for more information about generating verilog for your target FPGA.

If you have further questions, please consider posting on forums.sifive.com, as once this issue is closed others who have the same question probably won't see it.

closed 2016-07-28 by mwachs5