sifive/freedom-e-sdk #9: trying to build on mac `ld: library not found for -lrt`

nathantsoi, 2016-11-30:
should i just build on ubuntu or is there osx support?
```
gcc -DHAVE_CONFIG_H -DWITH_DEFAULT_MODEL='"RV32G"' -DWITH_ALIGNMENT=NONSTRICT_ALIGNMENT -DWITH_TARGET_WORD_BITSIZE=32 -DWITH_TARGET_BYTE_ORDER=BFD_ENDIAN_LITTLE   -DDEFAULT_INLINE=0   -Wall -Wdeclaration-after-statement -Wpointer-arith -Wpointer-sign -Wno-unused -Wunused-value -Wunused-function -Wno-switch -Wno-char-subscripts -Wmissing-prototypes -Wdeclaration-after-statement -Wempty-body -Wmissing-parameter-type -Wold-style-declaration -Wold-style-definition -Wformat-nonliteral      -I. -I/Users/me/src/freedom-e-sdk/riscv-gnu-toolchain/riscv-binutils-gdb/sim/riscv -I../common -I/Users/me/src/freedom-e-sdk/riscv-gnu-toolchain/riscv-binutils-gdb/sim/riscv/../common -I../../include -I/Users/me/src/freedom-e-sdk/riscv-gnu-toolchain/riscv-binutils-gdb/sim/riscv/../../include -I../../bfd -I/Users/me/src/freedom-e-sdk/riscv-gnu-toolchain/riscv-binutils-gdb/sim/riscv/../../bfd -I../../opcodes -I/Users/me/src/freedom-e-sdk/riscv-gnu-toolchain/riscv-binutils-gdb/sim/riscv/../../opcodes -I../../intl -g -O2 -Wl,-no_pie  -o run \
	  nrun.o libsim.a ../../bfd/libbfd.a ../../opcodes/libopcodes.a ../../intl/libintl.a -liconv ../../libiberty/libiberty.a -L../../zlib -lz -lrt -ldl
ld: library not found for -lrt
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

mwachs5, 2016-11-30:
AT this time, the vast majority of our testing has been done on Linux. We know of several issues compiling for Mac OSX, all of which can be worked around, but if you have Ubuntu access you might be happier overall. But I'll look into the workaround for this issue and we can build up the Mac OSX support.

nathantsoi, 2016-12-01:
thanks for the response! just to follow up, ubuntu 16.04 worked great after installing deps: `sudo apt install -y build-essential texinfo gawk bison libmpc-dev libtool automake libusb-1.0-0-dev` feel free to close

closed 2016-12-01 by nathantsoi