sifive/freedom-e-sdk #10: Build error in openocd with recent gcc

tdaede, 2016-12-01:
openocd builds with -Werror, and this warning occurs with gcc 6.2.1:

```
In file included from /home/thomas/sandbox/freedom-e-sdk/openocd/src/target/riscv/riscv.c:12:0:
/home/thomas/sandbox/freedom-e-sdk/openocd/src/target/riscv/riscv.c: In function ‘riscv_examine’:
/home/thomas/sandbox/freedom-e-sdk/openocd/src/target/riscv/riscv.c:61:69: error: left shift of negative value [-Werror=shift-negative-value]
 #define get_field(reg, mask) (((reg) & (mask)) / ((mask) & ~((mask) << 1)))
```

timsifive, 2016-12-01:
I don't have easy access to gcc 6. Can you paste the full output of the build command?

tdaede, 2016-12-01:
The simple solution is just to patch Makefile (this will prevent accidental breakage from future gcc versions too):

```
-       $(openocd_srcdir)/configure --prefix=$(openocd_dest)
+       $(openocd_srcdir)/configure --prefix=$(openocd_dest) --disable-werror

```

The patch to openocd will be a bit bigger as it looks like it's using signed types when it should be using unsigned.

Attached is the whole output before the patch.
[error.txt](https://github.com/sifive/freedom-e-sdk/files/625943/error.txt)

timsifive, 2016-12-01:
This is also https://github.com/sifive/openocd/issues/6

timsifive, 2016-12-01:
Is the patch to general parts of OpenOCD, or just the riscv portion? I'd like to fix the riscv portion regardless. I'll have to set up gcc 6 so I can quickly iterate.

tdaede, 2016-12-01:
The only instance of this warning is in the riscv portion, and I can confirm that the linked patch fixes the warning and allows the build to complete successfully.

timsifive, 2016-12-02:
Thanks. I just pushed that patch.

Tim

mwachs5, 2016-12-05:
I'll bump OpenOCD version and close this issue once that's tested and pushed to master.

timsifive, 2016-12-05:
On Mon, Dec 5, 2016 at 8:49 AM, Megan Wachs <notifications@github.com>
wrote:

> I'll bump OpenOCD version and close this issue once that's tested and
> pushed to master.
>
It was tested, and pushed to "master." In this repo "riscv" is the master
branch.

Tim

mwachs5, 2016-12-05:
Sorry, I mean the master branch of freedom-e-sdk. I believe it's still pointing to a slightly older OpenOCD.

PR #16 (2016-12-06): bump OpenOCD to fix compile warnings
Fixes #10
closed 2016-12-06 by mwachs5