sifive/freedom-e-sdk #15: Should include 'C' extension in the Makefile

mwachs5, 2016-12-05:
The current FE300 systems support the 'C' compressed extension, but the Makefile doesn't use it. Modify the Makefile to use compressed extension.

timsifive, 2016-12-05:
I don't think gdb knows how to set breakpoints in compressed code yet.

Tim

timsifive, 2016-12-05:
For reference, this hack “fixes” it but will break gdb on cores that don’t
support the C extension:

diff --git a/gdb/riscv-tdep.c b/gdb/riscv-tdep.c
index d788e33..3b56b9a 100644
--- a/gdb/riscv-tdep.c
+++ b/gdb/riscv-tdep.c
@@ -165,9 +165,13 @@ riscv_breakpoint_from_pc (struct gdbarch *gdbarch,
 {
   /* TODO: Support C.EBREAK for compressed (16-bit) insns.  */
   /* TODO: Support NOPs for >=6 byte insns.  */
-  static const gdb_byte sbreak_insn[] = { 0x73, 0x00, 0x10, 0x00, };

-  *bp_size = 4;
+
+  // static const gdb_byte sbreak_insn[] = { 0x73, 0x00, 0x10, 0x00, };
+  // *bp_size = 4;
+
+  static const gdb_byte sbreak_insn[] = { 0x02, 0x90 };
+  *bp_size = 2;

   return sbreak_insn;
 }

Tim
​

hirooih, 2016-12-24:
Thank you for the fix.

It seems that there is a better fix on the upstream.
https://github.com/riscv/riscv-binutils-gdb/blob/riscv-next/gdb/riscv-tdep.c

timsifive, 2016-12-24:
Yes, the latest gdb now has proper support for debugging compressed code. I'm still waiting to commit the relevant tests since for some reason I haven't been able to build a working set of tools lately.

mwachs5, 2017-01-26:
@timsifive I've tried bumping to the version of GDB as noted above, but it still doesn't seem to work if breakpoints are set at unaligned addresses. Does this need corresponding support in OpenOCD?

timsifive, 2017-01-26:
It shouldn't. Can you confirm that your gdb has code to read misa here: https://github.com/riscv/riscv-binutils-gdb/blob/riscv-next/gdb/riscv-tdep.c#L161

mwachs5, 2017-01-26:
> next/gdb/riscv-tdep.c#L161

Actually it does work, I was just using an out-of-date elf file. Sorry for
the false alarm.

mwachs5, 2017-01-26:
The GDB Version is bumped now and should work. "C" extension was already enabled by default a while ago.

closed 2017-01-26 by mwachs5