sifive/freedom-e-sdk #19: demo_gpio: mtvec set incorrectly to handle_trap

mwachs5, 2016-12-09:
The demo as-is doesn't handle the interrupts (timer or external) correctly.

During `_init`, we set `MTVEC` to `handle_trap`. But `handle_trap` is a C function which expects various things to be in registers and on the stack. We need an assembly level wrapper around this (which I believe we had in previous versions of the demo). I'll fix this.

aswaterman, 2016-12-09:
We might consider making the assembly wrapper (and MTVEC initialization) part of the standard crt0.

For people who don't want to provide a handle_trap, we can add something like this in crt0:

      .weak handle_trap
    handle_trap:
      j handle_trap

Since the symbol is weak, it will be overridden if the user provides their own implementation, but will avoid causing a link error if the user does not provide one.

mwachs5, 2016-12-10:
I renamed entry.S to start.S, then created entry.S which just handles the trap entry.

closed 2016-12-10 by mwachs5