sifive/freedom-e-sdk #35: Make plic_driver universally accessible

dgrubb, 2016-12-29:
Currently, plic_driver.c and plic_driver.h are only included as a part of the demo_gpio demo application.  The functionality of these files is useful beyond that application and will probably be reimplemented with new additions to software/, both by SiFive and third-party users of the SDK.

A precedent worth mimicking can found in the RIOT-OS RTOS [1] code structure where drivers for third-party accessories and internal SoC devices are held in top-level directories: 'drivers' and 'sys', respectively. They are then included through a Makefile reference allowing each application to share them in a reusable fashion.

Thanks.

[1] https://github.com/RIOT-OS/RIOT/

mwachs5, 2017-01-10:
I agree with this issue. Our "cinco" Arduino IDE code is also currently duplicating these same files, which should be avoided.

mwachs5, 2017-02-02:
closed by #50. Also we added a PRCI driver.

closed 2017-02-02 by mwachs5