sifive/freedom-e-sdk #36: Packages needed by Fedora 25 should be added to README,md

dpreed, 2017-01-01:
Since Ubuntu may not be the only system used, I suggest that a Fedora 25 package list be maintained as well.  The following should be included in that list:
texinfo mpfr mpfr-devel mpc libmpc-devel libtool gawk bison flex libusb-devel

mwachs5, 2017-01-10:
Can you provide the complete command a user would need to execute to install those packages on Fedora?

aaronscientiae, 2017-02-21:
I'm running Fedora 25 myself and got the compilation error. The below command installs the packages @dpreed listed with the addition of zlib-devel which I needed. I ran this and then the 'make tools' command compiled correctly.
`sudo dnf install -y autoconf automake texinfo mpfr mpfr-devel mpc libmpc-devel gmp libtool gawk bison flex libusb-devel zlib-devel`