sifive/freedom-e-sdk #51: Tools build problem on aarch64

laanwj, 2017-02-23:
Currently, `make tools` doesn't complete on aarch64. The platform is not recognized by `openocd`.

It was very simple to fix this: replace `configure.guess`  and `configure.sub` in `openocd/jimtcl/autosetup` with recent versions from GNU. E .g.
```bash
cd openocd/jimtcl/autosetup
wget -O config.guess 'http://git.savannah.gnu.org/gitweb/?p=config.git;a=blob_plain;f=config.guess;hb=HEAD'
wget -O config.sub 'http://git.savannah.gnu.org/gitweb/?p=config.git;a=blob_plain;f=config.sub;hb=HEAD'
```
This fix should probably be taken upstream, but I'm not quite sure where, and how to make the change propagate here, as it is three sub projects deep. Maybe [here](https://github.com/msteveb/autosetup)?

mwachs5, 2017-02-23:
Thanks for reporting this. It is probably best to file an issue against http://github.com/sifive/openocd.git, then we can bump this version once it is updated there.

laanwj, 2017-02-24:
Good point, I'll file it there too

mintisan, 2017-05-01:
It works for me on Jetson X2. Thanks.

closed 2019-03-26 by nategraff-sifive