sifive/freedom-e-sdk #52: SPI example

laanwj, 2017-02-26:
Right now, there is no example of using the SPI functionality of the board. It is not entirely trivial to figure it out using the documentation only.

FYI I've created [a demo that drives a small oled display from the HiFive board through SPI](https://github.com/laanwj/pmodoled) that may be useful as example...

Could file this as a pull request but I think it is too specific. I had the PMOD laying around from past FPGA experiments. Also from #41 I understand the plan is to make a separated demo repository.

mwachs5, 2017-02-27:
Very nice!

You're right, a good location for demo code is still on our TODO list. I'll keep this issue open until we come up with a better solution.

mwachs5, 2017-05-26:
Hi @laanwj , we have created a Wiki page for this repo where we hope people will post links to demo projects. I think that would be a good place for your SPI demo.

closed 2019-03-26 by nategraff-sifive