sifive/freedom-e-sdk #68: Not  "ARM-USB-TINY-H" debuger,use sdk commands uploading programs to Arty 7 encountered some problems

Pengnian, 2017-07-01:
Hi ,all

I'm  not use ARM-USB-TINY-H debuger, my jtag cable is "Xilinx platform cable USB". So,when I want to uploading program to my board(ARTY 7) by freedom-e-sdk commands, encountered some troubles. How to show the result of  executing program in my board .
[image]

[image]

timsifive, 2017-07-03:
You would have to change the config file to whatever configuration is required to have OpenOCD use your Xilinx Platform Cable USB. Googling suggests some people have made this work, but all the references are old and some mention special kernel modules.

You could try to figure that out, but I suspect it's more efficient to just by an ARM-USB-TINY-H.

Given the OpenOCD error messages, I do not believe the message that says the upload was successful.

mwachs5, 2017-07-03:
Yes, the "successful" error message is misleading. We need to fix the Makefile target which prints that.

Pengnian, 2017-07-10:
@timsifive Hi,Tim

Thanks your advice, I have brought the "ARM-USB-TINY-H" debugger, but situation not any change!
[image]

Pengnian, 2017-07-10:
@timsifive Thanks your advice, I have brought the "ARM-USB-TINY-H" debugger, but situation not any change! just use command of `make upload PROGRAM=hello BOARD=freedom-e300-arty` on freedom-e-sdk.

[image]

timsifive, 2017-07-10:
This is actually a big improvement. Previously OpenOCD was not able to talk
to your JTAG adapter. Now it is able to talk to the adapter, but when it
tries to talk to the RISC-V core something is going wrong.
This could be as simple as the board not being powered on or not programmed
with the right FPGA image, but I think the most likely problem is the
wiring between the Olimex adapter and the Arty board.

Tim

Pengnian, 2017-07-11:
@timsifive  Hi,Tim
At first ,thank your advice! The LED of debugger is light when connection between board and computer .
But when I use command "make upload PROGRAM=hello BOARD=freedom-e300-arty",the led not bright ! But every time the errors maybe difference!

[image]

timsifive, 2017-07-14:
The Makefile right now is making things more confusing. It looks like
OpenOCD starts, but then gdb can’t connect for some reason. I’d try running
things manual. So start OpenOCD in one shell (openocd -f
bsp/env/freedom-e300-arty/openocd.cfg), and then run gdb in another shell (
gdb, then target remote localhost:3333).

This may just be a timing issue where the Makefile starts gdb before
OpenOCD has fully started up.

Tim
​

closed 2019-03-26 by nategraff-sifive