sifive/freedom-e-sdk #83: freedom-e-sdk/riscv-gnu-toolchain/riscv-binutils-gdb/gdb/guile/scm-ports.c:146:10: error: ‘scm_new_port_table_entry’ was not declared in this scope

devurandom, 2017-12-23:
I am trying to build the SDK using `make tools BOARD=freedom-e300-hifive1` as described in the readme, but the build fails:
```
[...]
g++ -x c++  -g -O2   -I. -I$BASEDIR/freedom-e-sdk/riscv-gnu-toolchain/riscv-binutils-gdb/gdb -I$BASEDIR/freedom-e-sdk/riscv-gnu-toolchain/riscv-binutils-gdb/gdb/common -I$BASEDIR/freedom-e-sdk/riscv-gnu-t
oolchain/riscv-binutils-gdb/gdb/config -DLOCALEDIR="\"$BASEDIR/freedom-e-sdk/work/build/riscv-gnu-toolchain/riscv64-unknown-elf/prefix/share/locale\"" -DHAVE_CONFIG_H -I$BASEDIR/freedom-e-sdk/riscv-gnu-toolchain/riscv-binutils-
gdb/gdb/../include/opcode -I$BASEDIR/freedom-e-sdk/riscv-gnu-toolchain/riscv-binutils-gdb/gdb/../opcodes/.. -I$BASEDIR/freedom-e-sdk/riscv-gnu-toolchain/riscv-binutils-gdb/gdb/../readline/.. -I$BASEDIR/fr
eedom-e-sdk/riscv-gnu-toolchain/riscv-binutils-gdb/gdb/../zlib -I../bfd -I$BASEDIR/freedom-e-sdk/riscv-gnu-toolchain/riscv-binutils-gdb/gdb/../bfd -I$BASEDIR/freedom-e-sdk/riscv-gnu-toolchain/riscv-binutils-gdb/gdb/../include -
I../libdecnumber -I$BASEDIR/freedom-e-sdk/riscv-gnu-toolchain/riscv-binutils-gdb/gdb/../libdecnumber  -I$BASEDIR/freedom-e-sdk/riscv-gnu-toolchain/riscv-binutils-gdb/gdb/gnulib/import -Ibuild-gnulib/import   -DTUI=1   -pthread 
-I/usr/include/guile/2.2 -I/usr/include/python3.6m -I/usr/include/python3.6m -Wall -Wpointer-arith -Wno-unused -Wunused-value -Wunused-function -Wno-switch -Wno-char-subscripts -Wempty-body -Wunused-but-set-parameter -Wunused-but-set-variable -Wno-sign-compare -Wno-narrowi
ng -Wno-error=maybe-uninitialized  -c -o scm-ports.o -MT scm-ports.o -MMD -MP -MF .deps/scm-ports.Tpo $BASEDIR/freedom-e-sdk/riscv-gnu-toolchain/riscv-binutils-gdb/gdb/guile/scm-ports.c                                                                 
$BASEDIR/freedom-e-sdk/riscv-gnu-toolchain/riscv-binutils-gdb/gdb/guile/scm-ports.c: In function ‘scm_unused_struct* ioscm_open_port(scm_t_bits, long int)’:                                                                                              
$BASEDIR/freedom-e-sdk/riscv-gnu-toolchain/riscv-binutils-gdb/gdb/guile/scm-ports.c:146:10: error: ‘scm_new_port_table_entry’ was not declared in this scope                                                                                              
   port = scm_new_port_table_entry (port_type);                                                                                                                                                                                                                                  
          ^~~~~~~~~~~~~~~~~~~~~~~~                                                                                                                                                                                                                                               
$BASEDIR/freedom-e-sdk/riscv-gnu-toolchain/riscv-binutils-gdb/gdb/guile/scm-ports.c:146:10: note: suggested alternative: ‘scm_c_weak_table_put_x’                                                                                                         
   port = scm_new_port_table_entry (port_type);                                                                                                                                                                                                                                  
          ^~~~~~~~~~~~~~~~~~~~~~~~                                                                                                                                                                                                                                               
          scm_c_weak_table_put_x                                                                                                                                                                                                                                                 
$BASEDIR/freedom-e-sdk/riscv-gnu-toolchain/riscv-binutils-gdb/gdb/guile/scm-ports.c: In function ‘int ioscm_fill_input(SCM)’:                                                                                                                             
$BASEDIR/freedom-e-sdk/riscv-gnu-toolchain/riscv-binutils-gdb/gdb/guile/scm-ports.c:231:20: error: ‘SCM_PTAB_ENTRY’ was not declared in this scope                                                                                                        
   scm_t_port *pt = SCM_PTAB_ENTRY (port);                                                                                                                                                                                                                                       
                    ^~~~~~~~~~~~~~                                                                                                                                                                                                                                               
$BASEDIR/freedom-e-sdk/riscv-gnu-toolchain/riscv-binutils-gdb/gdb/guile/scm-ports.c:231:20: note: suggested alternative: ‘SCM_VTABLE_NAME’                                                                                                                
   scm_t_port *pt = SCM_PTAB_ENTRY (port);                                                                                                                                                                                                                                       
                    ^~~~~~~~~~~~~~                                                                                                                                                                                                                                               
                    SCM_VTABLE_NAME
[...]
```

I am using Arch Linux on this machine. Guile 2.2.3 is installed.

jim-wilson, 2017-12-23:
New versions of guile dropped support for some features that gdb uses.  Eventually gdb and guile will get in sync again.  Meanwhile, the solution is to disable guile when configuring gdb or install an older guile version that will work with gdb.  There was a patch added to fix this in https://github.com/riscv/riscv-gnu-toolchain/pull/247 that adds a --with-guile=X option.  You can use --with-guile=no to disable guile.  Or you  pass a version number to --with-guile and configure will try to use pkg-config to find it.  Just disabling it is simpler, as I think most people aren't using guile.

ilg-ul, 2017-12-23:
> install an older guile version that will work with gdb

any idea which is the latest functional version?

jim-wilson, 2017-12-23:
Looks like gdb developers choose to stop using guile-2.2.  See for instance
https://sourceware.org/bugzilla/show_bug.cgi?id=21104
This was apparently not backported to the gdb-7 branch.

I see a report that guile 2.1.6 also fails, and reports that guile-2.0 works (and probably any 2.0.x).

ilg-ul, 2017-12-23:
Thank you, Jim.

2.1.6 might have been an internal version, since I can't see it on https://ftp.gnu.org/gnu/guile/.

The GNU MCU RISC-V GCC build environment compiles all required libraries from sources, regardless of the versions installed in the build environment, so, if needed, I guess I can add guile-2.0.14 to the build without major problems.

However I'm not sure it is really needed...

jim-wilson, 2017-12-23:
I know that the python scripting support is in use, riscv-gcc/gcc/gdbhooks.py uses it for instance, but I don't know if anyone uses the guile scripting support.