sifive/freedom-e-sdk #85: Not able to download all sub-folders data in windows through ZIP file

thotaraghava, 2018-01-05:
Hi Everybody,

      I wanted to install risc-v-toolchain to work on PULPino simulations.
      I have downloaded ZIP file from github and could see sub folders data missing.
      Can you tell me how to download entire database.

     Your help is highly appreciated.

Thanks,

ilg-ul, 2018-01-05:
I guess those subfolders are git submodules.

try to clone the repository and be sure you recurse on submodules.

thotaraghava, 2018-01-05:
Yes, they are git sub modules.
Please tell me how to recurse on submodules while downloading onto windows.
Thanks for responding.

thotaraghava, 2018-01-05:
How to do this command while downloading ZIP file in windows.

git clone --recursive <URL-OF-REPOSITORY>

ilg-ul, 2018-01-05:
ignore the ZIPs, you need to clone the repository.

thotaraghava, 2018-01-05:
Should I download GitHub Desktop for this.

ilg-ul, 2018-01-05:
nope, git is enough.

thotaraghava, 2018-01-05:
[image]
 Which option I need to take. Open in Desktop is asking to download GitHub Desktop, the other option left is ZIP file.

thotaraghava, 2018-01-05:
Yes, GitHub Desktop looks working. I have installed and trying to clone, process is going on.
Thanks a lot for your patience ilg-ul.

ilg-ul, 2018-01-05:
BTW, after you manage to clone the repository, all you can do is examine the files.

if you want to build the samples, you also need the toolchain, a functional make and possibly more.

thotaraghava, 2018-01-05:
ok, sure.

thotaraghava, 2018-01-06:
HI,
 In the path /freedom-e-sdk/riscv-gnu-toolchain/riscv-binutils-gdb/, I dont see any configure file both in web page as well as in cloned database.
But while running make tools [BOARD=freedom-e300-hifive1], I am seeing below error. 

freedom-e-sdk/riscv-gnu-toolchain/riscv-binutils-gdb/configure: No such file or directory.

Please help.

ilg-ul, 2018-01-06:
the path of building the toolchain from sources requires a Linux system and some good Linux knowledge.

perhaps you should try to use the already made binaries.

alternatively you can use Eclipse, it is easier to use for first time users. in the latest version of GNU MCU Eclipse you have a SiFive project template that generates ready to run HiFive1 blinky projects.

thotaraghava, 2018-01-06:
Already made binaries means, the database on which somebody did already right

ilg-ul, 2018-01-06:
it means ready to run executables, not source code programs that need to be compiled, like you are trying to do.

thotaraghava, 2018-01-06:
ok

closed 2018-01-10 by thotaraghava