sifive/freedom-e-sdk #91: feedom-e-sdk upload demo occur errors

cailinlin5542, 2018-04-09:
When I upload demo to fpga, there are some errors, as follow:

cailinlin@cailinlin-Lenovo-Invalid-entry-length-16-Fixed-up-to-11:~/RISCV/freedom-e-sdk$ make upload [PROGRAM=demo_gpio] [BOARD=coreplexip-e31-arty]
work/build/openocd/prefix/bin/openocd -f bsp/env/coreplexip-e31-arty/openocd.cfg & \
/home/cailinlin/RISCV/freedom-e-sdk/work/build/riscv-gnu-toolchain/riscv64-unknown-elf/prefix/bin/riscv64-unknown-elf-gdb software/hello/hello --batch -ex "set remotetimeout 240" -ex "target extended-remote localhost:3333" -ex "monitor reset halt" -ex "monitor flash protect 0 64 last off" -ex "load" -ex "monitor resume" -ex "monitor shutdown" -ex "quit" && \
echo "Successfully uploaded 'hello' to coreplexip-e31-arty."
Open On-Chip Debugger 0.10.0+dev (2018-03-22-10:28)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 10000 kHz
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 10000 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0x20000913 (mfg: 0x489 (SiFive, Inc.), part: 0x0000, ver: 0x2)
Info : dtmcontrol_idle=5, dmi_busy_delay=1, ac_busy_delay=0
Info : dtmcontrol_idle=5, dmi_busy_delay=2, ac_busy_delay=0
Info : dtmcontrol_idle=5, dmi_busy_delay=3, ac_busy_delay=0
Info : dtmcontrol_idle=5, dmi_busy_delay=4, ac_busy_delay=0
Info : Disabling abstract command reads from CSRs.
Info : Disabling abstract command writes to CSRs.
Info : [0] Found 2 triggers
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, 2 triggers
Info : Listening on port 3333 for gdb connections
Ready for Remote Connections
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : accepting 'gdb' connection on tcp/3333
Error: FESPI_WRITE_REG error
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected
Remote connection closed
"monitor" command not supported by this target.
"monitor" command not supported by this target.
You can't do that when your target is `exec'
"monitor" command not supported by this target.
"monitor" command not supported by this target.
Successfully uploaded 'hello' to coreplexip-e31-arty.

I do not know how to solve this problem, Can you help me to solve it?
Thanks!

mwachs5, 2018-04-09:
This looks like you are using the wrong `cfg` file for your target. How did you obtain/generate the MCS/BIT file that you put on your FPGA? 

The `.cfg` file you are using is designed to work with the MCS file for E31 Core IP evaluation available from https://dev.sifive.com/risc-v-core-ip/evaluate/fpga/

If that's not where your image came from, you need to fix your `.cfg` file to remove or correct the `flash bank` command.

cailinlin5542, 2018-04-10:
Thanks for your reply!
I want to implement on Nexys4 DDR board,  I modify some FPGA configuration file based on Freedom code of Arty board and generate bit file.

1.
.cfg file I fix expected-id as follow:
jtag newtap $_CHIPNAME cpu -irlen 5 -expected-id 0x20000913
if I don't fix expected-id there are some errors as follow:
2.
flash bank I don't fix and it same as Arty board as follow:
flash bank spi0 fespi 0x40000000 0 0 0 $_TARGETNAME.0 0x20004000

Open On-Chip Debugger 0.10.0+dev-g2c183e3-dirty (2018-01-03-10:25)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 10000 kHz
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 10000 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0x20000913 (mfg: 0x489 (SiFive, Inc.), part: 0x0000, ver: 0x2)
Warn : JTAG tap: riscv.cpu       UNEXPECTED: 0x20000913 (mfg: 0x489 (SiFive, Inc.), part: 0x0000, ver: 0x2)
Error: JTAG tap: riscv.cpu  expected 1 of 1: 0x20000001 (mfg: 0x000 (<invalid>), part: 0x0000, ver: 0x2)
Error: Trying to use configured scan chain anyway...
Warn : Bypassing JTAG setup events due to errors
Info : dtmcontrol_idle=5, dmi_busy_delay=1, ac_busy_delay=0
Info : dtmcontrol_idle=5, dmi_busy_delay=2, ac_busy_delay=0
Info : dtmcontrol_idle=5, dmi_busy_delay=3, ac_busy_delay=0
Info : dtmcontrol_idle=5, dmi_busy_delay=4, ac_busy_delay=0
Info : Disabling abstract command reads from CSRs.
Info : Disabling abstract command writes to CSRs.
Info : [0] Found 2 triggers
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, 2 triggers
Info : Listening on port 3333 for gdb connections
Error: FESPI_WRITE_REG error
Error: auto_probe failed

And I don't know how to fix this command, can you give some suggestions?
Thanks!

closed 2019-03-26 by nategraff-sifive