sifive/freedom-e-sdk #92: Freedom port to Nexys4 DDR board

cailinlin5542, 2018-04-10:
Freedom only can run on Arty and VC707, but I want to run on Nexys4 DDR board. So I fix some fpga configuration file. After I fixing, I execute command "make -f Makefile.e300nexys4ddrdevkit mcs".  There are some errors as follow:

`Phase 1.2 IO Placement/ Clock Placement/ Build Placer Device
INFO: [Timing 38-35] Done setting XDC timing constraints.
CRITICAL WARNING: [Place 30-722] Terminal 'qspi_cs' has IOB constraint set to TRUE, but it is either not connected to a FLOP element or the connected FLOP element could not be brought into the I/O
ERROR: [Place 30-574] Poor placement for routing between an IO pin and BUFG. If this sub optimal condition is acceptable for this design, you may use the CLOCK_DEDICATED_ROUTE constraint in the .xdc file to demote this message to a WARNING. However, the use of this override is highly discouraged. These examples can be used directly in the .xdc file to override this clock rule.
	< set_property CLOCK_DEDICATED_ROUTE FALSE [get_nets IOBUF_6/O] >
	IOBUF_6/IBUF (IBUF.O) is locked to IOB_X1Y115
	IOBUF_6_O_BUFG_inst (BUFG.I) is provisionally placed by clockplacer on BUFGCTRL_X0Y31
Resolution: Poor placement of an IO pin and a BUFG has resulted in the router using a non-dedicated path between the two.  There are several things that could trigger this DRC, each of which can cause unpredictable clock insertion delays that result in poor timing.  This DRC could be caused by any of the following: (a) a clock port was placed on a pin that is not a CCIO-pin (b)the BUFG has not been placed in the same half of the device or SLR as the CCIO-pin (c) a single ended clock has been placed on the N-Side of a differential pair CCIO-pin.
Phase 1.2 IO Placement/ Clock Placement/ Build Placer Device | Checksum: ffcc549c
Time (s): cpu = 00:00:12 ; elapsed = 00:00:11 . Memory (MB): peak = 2962.973 ; gain = 0.000 ; free physical = 1683 ; free virtual = 3488
Phase 1 Placer Initialization | Checksum: ffcc549c
Time (s): cpu = 00:00:12 ; elapsed = 00:00:11 . Memory (MB): peak = 2962.973 ; gain = 0.000 ; free physical = 1683 ; free virtual = 3488
ERROR: [Place 30-99] Placer failed with error: 'IO Clock Placer failed'
Please review all ERROR, CRITICAL WARNING, and WARNING messages during placement to understand the cause for failure.
Ending Placer Task | Checksum: ffcc549c
Time (s): cpu = 00:00:12 ; elapsed = 00:00:12 . Memory (MB): peak = 2962.973 ; gain = 0.000 ; free physical = 1682 ; free virtual = 3488
INFO: [Common 17-83] Releasing license: Implementation
13 Infos, 0 Warnings, 1 Critical Warnings and 3 Errors encountered.
place_design failed
ERROR: [Common 17-69] Command failed: Placer could not place all instances
    while executing
"source [file join $scriptdir "place.tcl"]"
    (file "/home/RISCV/freedom/fpga-shells/xilinx/common/tcl/vivado.tcl" line 24)
INFO: [Common 17-206] Exiting Vivado at Tue Apr 10 11:04:11 2018...
common.mk:81: recipe for target '/home/cailinlin/RISCV/freedom/builds/e300nexys4ddrdevkit/obj/E300Nexys4DDRDevKitFPGAChip.bit' failed
make: *** [/home/RISCV/freedom/builds/e300nexys4ddrdevkit/obj/E300Nexys4DDRDevKitFPGAChip.bit] Error 1`

And I correct error according to tips, Add the 
"set_property CLOCK_DEDICATED_ROUTE FALSE [get_nets IOBUF_6/O]" 
in nexys4_ddr-master.xdc. And I execute command "make -f Makefile.e300nexys4ddrdevkit mcs" again. According to tips, I also add the 
"set_property SEVERITY {Warning} [get_drc_checks NSTD-1]
set_property SEVERITY {Warning} [get_drc_checks UCIO-1]"
in the same file. And there is no error, and generate the bit file.

All files which I fixed according to Arty files, If I fix error like above, whether  there will exist potential problem. Because when I upload demo occurs some problems.

mhanuel26, 2018-04-27:
did you solve the issue?, I am experience something similar to E300.

nategraff-sifive, 2019-03-26:
Closing, not related to Freedom E SDK

closed 2019-03-26 by nategraff-sifive