sifive/freedom-e-sdk #95: 8 bit access to CLIC

jeras, 2018-07-03:
I noticed 8 bit peripheral accesses (uint8_t) are used in CLIC driver.

https://github.com/sifive/freedom-e-sdk/blob/clic-vectored/bsp/drivers/clic/clic_driver.c

The standard bus width for peripherals on 32 bit and probably 64 bit systems is 32 bits. And it is unusual to access peripheral registers with cycles smaller then 32 bits (64 bit accesses could be split by the interconnect). Many (probably most most) peripherals do not have byte select signals, therefore all registers have to be accessed as a single unit. I usually configured the peripheral bus to return bus errors if smaller then 32 bit write accesses were made.

So I was surprised to find 8 bit accesses to memory mapped CLIC registers. Is this a part of an encoding scheme to make the address space smaller, so immediate offset instructions could be used? Or is there another reason?

jeras, 2018-07-04:
I checked the document again, and indeed 8 bit registers are used to fit 1024 interrupts into 4kB of memory (1024 * (intip, intie, intcfg registers) + cfg register + 1023 unused bytes).
https://github.com/sifive/clic-spec/blob/master/clic.adoc#m-mode-clic-regions

It is unexpected, so some extra care should be taken while designing the interconnect.

closed 2018-07-04 by jeras

solomatnikov, 2018-07-06:
TileLink supports 1-byte accesses for any bus width