sifive/freedom-e-sdk #100: OpenOCD does not work after first use

AndreasDa, 2018-08-28:
Hello, I'm trying to debug some program on the freedom e300 arty board using openocd.
First, I uploaded the demo program demo_gpio with:
make upload PROGRAM=demo_gpio BOARD=freedom-e300-arty
After that I did:
make run_openocd
make run gdb
in two separate terminal windows.
After connecting the gdb to the server provided by openocd, I tried to do one "(gdb) next" operation and got an infinite number of:
program halted at <some address>
Upon that, i closed the windows to stop the infinite loop.
Now, I can neither upload new programs, nor debug the program already on the board. 
This is the error message I get when trying to run openocd with make run_openocd:
Error: no device found
Error: unable to open ftdi device with vid 0403, pid 6010, description 'Dual RS232-HS', serial '*' at bus location '*'
Restarting my computer, disconnecting and reconnecting the board and resetting the board did not help.
Any idea how to resolve this issue would be greatly appreciated.

emfive, 2018-08-28:
Is the light on the Olimex Programmer on?

AndreasDa, 2018-08-29:
Yes.
The problem was that not the ftdi of the olimex was used, but the one on the arty board due to wrong configuration. This lead to the error above.

closed 2018-09-06 by AndreasDa