sifive/freedom-e-sdk #145: Permission denied (publickey)

hyf6661669, 2019-01-14:
It seems that the permission setting is not corrected:
```
$ git submodule update --init --recursive
Cloning into '/home/hyf/freedom-e-sdk/software/local-interrupt'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:sifive/example-local-interrupt.git' into submodule path '/home/hyf/freedom-e-sdk/software/local-interrupt' failed
Failed to clone 'software/local-interrupt'. Retry scheduled
Cloning into '/home/hyf/freedom-e-sdk/software/software-interrupt'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:sifive/example-software-interrupt.git' into submodule path '/home/hyf/freedom-e-sdk/software/software-interrupt' failed
Failed to clone 'software/software-interrupt'. Retry scheduled
Cloning into '/home/hyf/freedom-e-sdk/software/timer-interrupt'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:sifive/example-timer-interrupt.git' into submodule path '/home/hyf/freedom-e-sdk/software/timer-interrupt' failed
Failed to clone 'software/timer-interrupt'. Retry scheduled
Cloning into '/home/hyf/freedom-e-sdk/software/local-interrupt'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:sifive/example-local-interrupt.git' into submodule path '/home/hyf/freedom-e-sdk/software/local-interrupt' failed
Failed to clone 'software/local-interrupt' a second time, aborting
```

lcbcFoo, 2019-01-14:
I second this, same output if directly cloning with submodules:
`git clone --recursive https://github.com/sifive/freedom-e-sdk.git`

jim-wilson, 2019-01-14:
This is a problem with the .gitmodules file, which has
[submodule "software/software-interrupt"]
	path = software/software-interrupt
	url = git@github.com:sifive/example-software-interrupt.git
[submodule "software/timer-interrupt"]
	path = software/timer-interrupt
	url = git@github.com:sifive/example-timer-interrupt.git
[submodule "software/local-interrupt"]
	path = software/local-interrupt
	url = git@github.com:sifive/example-local-interrupt.git

If you rewrite those URLs to use the https form instead of the ssh form, then it should work.  Since everyone inside SiFive does have ssh access, this will work for all of us and we won't see the problem unless we use an unprivileged account to try to reproduce.

lcbcFoo, 2019-01-14:
Just tried again and worked perfectly for me, thanks Jim and bsousi. =)

closed 2019-01-14 by jim-wilson