sifive/freedom-e-sdk #178: configure in freedom-metal doesn't respect `RISCV_PATH`

palmer-dabbelt, 2019-02-22:
I'm not sure exactly what's going on here, but if I set `RISCV_PATH` but don't have a compiler in `PATH` then the run of `configure` in `freedom-metal` doesn't pick up the target GCC.

bsousi5, 2019-02-22:
Wouldnt that expected, since configure script would abort because there is no GCC, like here,
checking for gcc... gcc
checking whether the C compiler works... no
configure: error: in `/foo/bsp/bar/build':
configure: error: C compiler cannot create executables
See `config.log' for more details
scripts/libmetal.mk:13: recipe for target ‘/foo/bsp//bar/build/Makefile' failed

Or May be I am missing an expectation?

jim-wilson, 2019-02-22:
You could write some configure tests for this, but the usual way is to insist that people set their path correctly first.  If you want configure tests, you might be able to use AC_PATH_PROG.
https://www.gnu.org/software/autoconf/manual/autoconf.html#Generic-Programs
This doesn't set PATH, it just sets a variable that you can use in later rules, but you could use this to look for the cross compiler and other tools.  Or I suppose you could just set PATH at the top of the configure script to include RISCV_PATH, but then you will need another step to get it into the Makefile somehow.

nategraff-sifive, 2019-03-27:
I wasn't able to reproduce this, at least not in a Docker container
```
root@6f0ed64de951:/freedom-e-sdk# echo $PATH
/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin

root@6f0ed64de951:/freedom-e-sdk# echo $RISCV_PATH
/toolchain/riscv64-unknown-elf-gcc-8.1.0-2018.12.0-x86_64-linux-ubuntu14

root@6f0ed64de951:/freedom-e-sdk# make software
cd /freedom-e-sdk/bsp/sifive-hifive1/build/debug/ && \
	CFLAGS="-march=rv32imac -mabi=ilp32 -mcmodel=medlow -ffunction-sections -fdata-sections -I/freedom-e-sdk/bsp/sifive-hifive1/install/include -O0 -g" \
	/freedom-e-sdk/freedom-metal/configure \
	--host=riscv64-unknown-elf \
	--prefix=/freedom-e-sdk/bsp/sifive-hifive1/install \
	--libdir=/freedom-e-sdk/bsp/sifive-hifive1/install/lib/debug \
	--disable-maintainer-mode \
	--with-preconfigured \
	--with-machine-name=sifive-hifive1 \
	--with-machine-header=/freedom-e-sdk/bsp/sifive-hifive1/metal.h \
	--with-machine-ldscript=/freedom-e-sdk/bsp/sifive-hifive1/metal.default.lds \
	--with-builtin-libgloss
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for riscv64-unknown-elf-strip... riscv64-unknown-elf-strip
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether to enable maintainer-specific portions of Makefiles... no
checking for riscv64-unknown-elf-gcc... riscv64-unknown-elf-gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... yes
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether riscv64-unknown-elf-gcc accepts -g... yes
checking for riscv64-unknown-elf-gcc option to accept ISO C89... none needed
checking whether riscv64-unknown-elf-gcc understands -c and -o together... yes
checking for style of include used by make... GNU
checking dependency style of riscv64-unknown-elf-gcc... gcc3
checking for riscv64-unknown-elf-ranlib... riscv64-unknown-elf-ranlib
checking for riscv64-unknown-elf-ar... riscv64-unknown-elf-ar
checking the archiver (riscv64-unknown-elf-ar) interface... ar
checking dependency style of riscv64-unknown-elf-gcc... gcc3
checking whether C compiler accepts mmachine=... no
checking whether C compiler accepts menv=metal... no
checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: creating Makefile
config.status: executing depfiles commands
touch -c /freedom-e-sdk/bsp/sifive-hifive1/build/debug/Makefile
...
```

docandrew, 2019-04-14:
> 
> 
> I'm not sure exactly what's going on here, but if I set `RISCV_PATH` but don't have a compiler in `PATH` then the run of `configure` in `freedom-metal` doesn't pick up the target GCC.

I just ran into this same issue tonight - earlier in the config script it seems to pick up the related tools like riscv64-unknown-elf-gcc/bin/riscv64-unknown-elf-strip, so I'm not sure why it is having trouble finding the compiler at the same location.

```
## ----------- ##
## Core tests. ##
## ----------- ##

configure:1929: checking for a BSD-compatible install
configure:1997: result: /bin/install -c
configure:2008: checking whether build environment is sane
configure:2063: result: yes
configure:2122: checking for riscv64-unknown-elf-strip
configure:2138: found /home/doc/riscv/riscv64-unknown-elf-gcc/bin/riscv64-unknown-elf-strip
configure:2149: result: riscv64-unknown-elf-strip
configure:2214: checking for a thread-safe mkdir -p
configure:2253: result: /bin/mkdir -p
configure:2260: checking for gawk
configure:2276: found /bin/gawk
configure:2287: result: gawk
configure:2298: checking whether make sets $(MAKE)
configure:2320: result: yes
configure:2349: checking whether make supports nested variables
configure:2366: result: yes
configure:2493: checking whether to enable maintainer-specific portions of Makefiles
configure:2502: result: no
configure:2546: checking for riscv64-unknown-elf-gcc
configure:2562: found /home/doc/riscv/riscv64-unknown-elf-gcc/bin/riscv64-unknown-elf-gcc
configure:2573: result: riscv64-unknown-elf-gcc
configure:2842: checking for C compiler version
configure:2851: riscv64-unknown-elf-gcc --version >&5
/home/doc/riscv/freedom-e-sdk/freedom-metal/configure: 1: eval: riscv64-unknown-elf-gcc: not found
```

jim-wilson, 2019-04-14:
This looks like a different unrelated problem.

Looks like it did find riscv64-unknown-elf-gcc, but it won't run on your system.  Did you download a toolchain from somewhere?  Did you verify that it works by compiling a hello world program?  If you get a "not found" error every time you try to run it then you could have a glibc version incompatibility.  Try running the command "ldd /home/doc/riscv/riscv64-unknown-elf-gcc/bin/riscv64-unknown-elf-gcc" and check for errors.  What OS and version is running on the host and where exactly did you get the toolchain release?

docandrew, 2019-04-14:
Jim, thanks for the quick response - I downloaded the toolchain right from the SiFive website. Looking at it, I think you're right about the libc version, ldd shows a bunch of reloc errors. I'll see if I can clone the toolchain repo and doing a clean build on the host, and let you know if that solves the issue!

docandrew, 2019-04-14:
OK - underlying libc error on my part - a clean install on a fresh Ubuntu VM seems to work fine, at least for hello.c.