sifive/freedom-e-sdk #182: Demo_gpio on for freedom-e310-arty compile failure  (uart.h)

salmansheikh, 2019-02-25:
I have compiled and programmed the mcs file for the freedome-e310 arty board and programmed the FPGA. I am getting errors on the compile using the the following command? 

Shouldn't this compile out of the box if I haven't touched any of the files?

Salman

 make software BSP=metal PROGRAM=demo_gpio TARGET=freedom-e310-arty

Entering directory `/home/ssheikh/freedom-e-sdk/bsp/freedom-e310-arty/build'
riscv64-unknown-elf-gcc -DPACKAGE_NAME=\"freedom-metal\" -DPACKAGE_TARNAME=\"freedom-metal\" -DPACKAGE_VERSION=\"v0.1.2\" -DPACKAGE_STRING=\"freedom-metal\ v0.1.2\" -DPACKAGE_BUGREPORT=\"palmer@sifive.com\" -DPACKAGE_URL=\"\" -DPACKAGE=\"freedom-metal\" -DVERSION=\"v0.1.2\" -I. -I/home/ssheikh/freedom-e-sdk/freedom-metal    -lriscv_metal_ -D__METAL_MACHINE_HEADER=\"/home/ssheikh/freedom-e-sdk/bsp/freedom-e310-arty/metal.h\" -march=rv32imac -mabi=ilp32 -g -mcmodel=medany -MT src/drivers/libriscv__mmachine__freedom-e310-arty_a-sifive,fe310-g000,pll.o -MD -MP -MF src/drivers/.deps/libriscv__mmachine__freedom-e310-arty_a-sifive,fe310-g000,pll.Tpo -c -o src/drivers/libriscv__mmachine__freedom-e310-arty_a-sifive,fe310-g000,pll.o `test -f 'src/drivers/sifive,fe310-g000,pll.c' || echo '/home/ssheikh/freedom-e-sdk/freedom-metal/'`src/drivers/sifive,fe310-g000,pll.c
In file included from /home/ssheikh/freedom-e-sdk/freedom-metal/metal/drivers/sifive,uart0.h:11:0,
                 from /home/ssheikh/freedom-e-sdk/bsp/freedom-e310-arty/metal.h:34,
                 from /home/ssheikh/freedom-e-sdk/freedom-metal/metal/machine.h:8,
                 from /home/ssheikh/freedom-e-sdk/freedom-metal/src/drivers/sifive,fe310-g000,pll.c:7:
/home/ssheikh/freedom-e-sdk/freedom-metal/metal/uart.h: In function 'metal_uart_putc':
/home/ssheikh/freedom-e-sdk/freedom-metal/metal/uart.h:50:91: error: 'const struct metal_uart_vtable' has no member named '__sputc_r'
 inline int metal_uart_putc(struct metal_uart *uart, unsigned char c) { return uart->vtable->putc(uart, c); }
                                                                                           ^~
/home/ssheikh/freedom-e-sdk/freedom-metal/metal/uart.h: In function 'metal_uart_getc':
/home/ssheikh/freedom-e-sdk/freedom-metal/metal/uart.h:58:106: error: macro "getc" passed 2 arguments, but takes just 1
 inline int metal_uart_getc(struct metal_uart *uart, unsigned char *c) { return uart->vtable->getc(uart, c); }
                                                                                                          ^
/home/ssheikh/freedom-e-sdk/freedom-metal/metal/uart.h:58:92: warning: return makes integer from pointer without a cast [-Wint-conversion]
 inline int metal_uart_getc(struct metal_uart *uart, unsigned char *c) { return uart->vtable->getc(uart, c); }
                                                                                ~~~~~~~~~~~~^~~~~~
make[1]: *** [src/drivers/libriscv__mmachine__freedom-e310-arty_a-sifive,fe310-g000,pll.o] Error 1
make[1]: Leaving directory `/home/ssheikh/freedom-e-sdk/bsp/freedom-e310-arty/build'
make: *** [/home/ssheikh/freedom-e-sdk/bsp//freedom-e310-arty/install/stamp] Error 2

bsousi5, 2019-02-25:
Hi Salam,

Freedom-e-sdk is currently in a transition of getting converted to use freedom-metal.
Unfortunately, demo_gpio has not been converted yet.

For this example, please drop “BSP=metal”, and it should compile file.

Regards,
Bunnaroath

jim-wilson, 2019-02-25:
Or try using the v1_0 branch, which is the last commit before the rewrite started.

salmansheikh, 2019-02-26:
I built the hello program and tried running it as follows and for some reason the load failed after loading text and rodata and I never see any hello world printed on my minicom screen. What could the problem be? The hello program build just fine.

make upload PROGRAM=hello TARGET=freedom-e300-arty
openocd -f bsp/env/freedom-e300-arty/openocd.cfg & \
riscv64-unknown-elf-gdb software/hello//hello --batch -ex "set remotetimeout 240" -ex "target extended-remote localhost:3333" -ex "monitor reset halt" -ex "monitor flash protect 0 64 last off" -ex "load" -ex "monitor resume" -ex "monitor shutdown" -ex "quit" && \
echo "Successfully uploaded 'hello' to freedom-e300-arty."
Open On-Chip Debugger 0.10.0+dev-gc9d31b8 (2018-11-27-06:08)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 10000 kHz
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 10000 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0x20000913 (mfg: 0x489 (SiFive, Inc.), part: 0x0000, ver: 0x2)
Info : datacount=1 progbufsize=16
Info : Disabling abstract command reads from CSRs.
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, misa=0x40001105
Info : Listening on port 3333 for gdb connections
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : accepting 'gdb' connection on tcp/3333
Info : Found flash device 'micron n25q128' (ID 0x0018ba20)
warning: Target-supplied registers are not supported by the current architecture
0x00007fff20401a44 in ?? ()
Info : JTAG tap: riscv.cpu tap/device found: 0x20000913 (mfg: 0x489 (SiFive, Inc.), part: 0x0000, ver: 0x2)
JTAG tap: riscv.cpu tap/device found: 0x20000913 (mfg: 0x489 (SiFive, Inc.), part: 0x0000, ver: 0x2)
cleared protection for sectors 64 through 255 on flash bank 0
cleared protection for sectors 64 through 255 on flash bank 0
Loading section .text, size 0x1fa6 lma 0x100b0
Loading section .rodata, size 0x18 lma 0x12058
**Load failed**
Info : Disabling abstract command writes to CSRs.
Disabling abstract command writes to CSRs.
shutdown command invoked
shutdown command invoked
A debugging session is active.

	Inferior 1 [Remote target] will be detached.

Quit anyway? (y or n) [answered Y; input not from terminal]
Remote connection closed
Successfully uploaded 'hello' to freedom-e300-arty.

lcbcFoo, 2019-03-11:
I also ran into this problem. The load failed occurs because openOCD has an incorrect address for the SPI memory section and when it tries to load the program into flash it tries to write at a wrong address. 

There are 2 options: You are using the wrong mcs file or you are using the wrong BSP or TARGET for your mcs file, which may also be caused by the transitioning they are doing in the repository. At this time, both:
```
make BSP=metal PROGRAM=hello BOARD=freedom-e310-arty upload
make BSP=mee PROGRAM=hello BOARD=freedom-e310-arty upload
```
works for me.

salmansheikh, 2019-03-11:
What are those two addresses? I am not at my computer.  I will try that.
Thanks.

lcbcFoo, 2019-03-11:
I added more information in the previous comment, should answer your question. But I believe you should not need to change anything and only need to use correct make flags. =)

Another thing, always run `make clean` before using another target or the binary may not be compiled again possibly causing other problems.

closed 2019-03-27 by nategraff-sifive