sifive/freedom-e-sdk #222: not able to configure and make bare matal program

sathya281, 2019-03-28:
Dear Experts,
I tried compiling the timer-interrupt program in freedom-sdk. I encountered below problem.
The toolchain details are at the end.

sathya281:~/freedom-e-sdk$ **make PROGRAM=timer-interrupt TARGET=coreip-s51-arty CONFIGURATION=release software**
cd /home/sathya/freedom-e-sdk/bsp/coreip-s51-arty/build/release/ && \
	CFLAGS="-march=rv64imac -mabi=lp64 -mcmodel=medany -ffunction-sections -fdata-sections -I/home/sathya/freedom-e-sdk/bsp/coreip-s51-arty/install/include -Os" \
	/home/sathya/freedom-e-sdk/freedom-metal/configure \
	--host=riscv64-unknown-elf \
	--prefix=/home/sathya/freedom-e-sdk/bsp/coreip-s51-arty/install \
	--libdir=/home/sathya/freedom-e-sdk/bsp/coreip-s51-arty/install/lib/release \
	--disable-maintainer-mode \
	--with-preconfigured \
	--with-machine-name=coreip-s51-arty \
	--with-machine-header=/home/sathya/freedom-e-sdk/bsp/coreip-s51-arty/metal.h \
	--with-machine-ldscript=/home/sathya/freedom-e-sdk/bsp/coreip-s51-arty/metal.default.lds \
	--with-builtin-libgloss
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for riscv64-unknown-elf-strip... riscv64-unknown-elf-strip
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether to enable maintainer-specific portions of Makefiles... no
checking for riscv64-unknown-elf-gcc... riscv64-unknown-elf-gcc
checking whether the C compiler works... no
configure: error: in `/home/sathya/freedom-e-sdk/bsp/coreip-s51-arty/build/release':
configure: error: C compiler cannot create executables
See `config.log' for more details
scripts/libmetal.mk:17: **recipe for target '/home/sathya/freedom-e-sdk/bsp/coreip-s51-arty/build/release/Makefile' failed**
make: *** [/home/sathya/freedom-e-sdk/bsp/coreip-s51-arty/build/release/Makefile] Error 77

> 
On checking the config.log, I found that 
[config.log](https://github.com/sifive/freedom-e-sdk/files/3016520/config.log)

configure:1929: checking for a BSD-compatible install
configure:1997: result: /usr/bin/install -c
configure:2008: checking whether build environment is sane
configure:2063: result: yes
configure:2122: checking for riscv64-unknown-elf-strip
configure:2138: found /media/sathya/secondary_drive/tools/bin/riscv64-unknown-elf-strip
configure:2149: result: riscv64-unknown-elf-strip
configure:2214: checking for a thread-safe mkdir -p
configure:2253: result: /bin/mkdir -p
configure:2260: checking for gawk
configure:2276: found /usr/bin/gawk
configure:2287: result: gawk
configure:2298: checking whether make sets $(MAKE)
configure:2320: result: yes
configure:2349: checking whether make supports nested variables
configure:2366: result: yes
configure:2493: checking whether to enable maintainer-specific portions of Makefiles
configure:2502: result: no
configure:2546: checking for riscv64-unknown-elf-gcc
configure:2562: found /media/sathya/secondary_drive/tools/bin/riscv64-unknown-elf-gcc
configure:2573: result: riscv64-unknown-elf-gcc
configure:2842: checking for C compiler version
configure:2851: riscv64-unknown-elf-gcc --version >&5
riscv64-unknown-elf-gcc (GCC) 8.2.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

configure:2862: $? = 0
configure:2882: checking whether the C compiler works
configure:2904: riscv64-unknown-elf-gcc -march=rv64imac -mabi=lp64 -mcmodel=medany -ffunction-sections -fdata-sections -I/home/sathya/freedom-e-sdk/bsp/coreip-s51-arty/install/include -Os   conftest.c  >&5
**_/media/sathya/secondary_drive/tools/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /tmp/cctas67H.o: can't link soft-float modules with double-float modules_**
/media/sathya/secondary_drive/tools/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /tmp/cctas67H.o
collect2: error: ld returned 1 exit status
                                             

**Note:**
sathya@sathya-HP-EliteDesk-800-G1-TWR:~/freedom-e-sdk$ openocd -version
Open On-Chip Debugger 0.10.0+dev-00529-g9fac2de (2019-03-19-12:48)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
openocd: invalid option -- 'e'

sathya@sathya-HP-EliteDesk-800-G1-TWR:~/freedom-e-sdk$ riscv64-unknown-elf-gcc --version
riscv64-unknown-elf-gcc (GCC) 8.2.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

bsousi5, 2019-03-28:
Hi Sathya,

What OS are you running on?

Bunnaroath

sathya281, 2019-03-28:
ubuntu 16.04 LTS

jim-wilson, 2019-03-28:
Looks like you don't have a multilib enabled compiler.  If you use the compiler on our web site, this will work.  If you want to build your own compiler, you must specify --enable-multilib when configuring the toolchain.

sunqingyao, 2019-05-21:
The similar issue here. Error message

```
~/Developer/freedom-e-sdk master#v201903?
❯ make software
cd /Users/nalzok/Developer/freedom-e-sdk/bsp/sifive-hifive1/build/debug/ && \
		CFLAGS="-march=rv32imac -mabi=ilp32 -mcmodel=medlow -ffunction-sections -fdata-sections -I/Users/nalzok/Developer/freedom-e-sdk/bsp/sifive-hifive1/install/include -O0 -g" \
		/Users/nalzok/Developer/freedom-e-sdk/freedom-metal/configure \
		--host=riscv64-unknown-elf \
		--prefix=/Users/nalzok/Developer/freedom-e-sdk/bsp/sifive-hifive1/install \
		--libdir=/Users/nalzok/Developer/freedom-e-sdk/bsp/sifive-hifive1/install/lib/debug \
		--disable-maintainer-mode \
		--with-preconfigured \
		--with-machine-name=sifive-hifive1 \
		--with-machine-header=/Users/nalzok/Developer/freedom-e-sdk/bsp/sifive-hifive1/metal.h \
		--with-machine-ldscript=/Users/nalzok/Developer/freedom-e-sdk/bsp/sifive-hifive1/metal.default.lds \
		--with-builtin-libgloss
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for riscv64-unknown-elf-strip... riscv64-unknown-elf-strip
checking for a thread-safe mkdir -p... /Users/nalzok/Developer/freedom-e-sdk/freedom-metal/install-sh -c -d
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether to enable maintainer-specific portions of Makefiles... no
checking for riscv64-unknown-elf-gcc... riscv64-unknown-elf-gcc
checking whether the C compiler works... no
configure: error: in `/Users/nalzok/Developer/freedom-e-sdk/bsp/sifive-hifive1/build/debug':
configure: error: C compiler cannot create executables
See `config.log' for more details
make: *** [/Users/nalzok/Developer/freedom-e-sdk/bsp/sifive-hifive1/build/debug/Makefile] Error 77
```

Log file: [config.log](https://github.com/sifive/freedom-e-sdk/files/3201024/config.log)

The toolchain (GNU Embedded Toolchain — v2019.02.0) is downloaded from https://www.sifive.com/boards
```
~/Developer/freedom-e-sdk master#v201903?
❯ riscv64-unknown-elf-gcc --version
riscv64-unknown-elf-gcc (GCC) 8.3.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

jim-wilson, 2019-05-21:
configure:2904: riscv64-unknown-elf-gcc -march=rv32imac -mabi=ilp32 -mcmodel=medlow -ffunction-sections -fdata-sections -I/Users/nalzok/Developer/freedom-e-sdk/bsp/sifive-hifive1/install/include -O0 -g   conftest.c  >&5
/opt/riscv/lib/gcc/riscv64-unknown-elf/8.3.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv/lib/gcc/riscv64-unknown-elf/8.3.0/../../../../riscv64-unknown-elf/lib/crt0.o: ABI is incompatible with that of the selected emulation:

Looks like you have a compiler that is not multilib enabled.  freedom-e-sdk works best if you use the compiler on the SiFive website.  If you want to build your own compiler, then you need to use --enable-multilib when configuring it.  Also, if you use FSF or riscv compiler sources, then you may run into problems with missing CLIC support, which is an SiFive local feature, so this works best if you build from our toolchain sources.

sunqingyao, 2019-05-21:
@jim-wilson Could you please provide a direct download link? The one I used is `https://static.dev.sifive.com/dev-tools/riscv64-unknown-elf-gcc-8.2.0-2019.02.0-x86_64-apple-darwin.tar.gz`

nategraff-sifive, 2019-05-21:
@sunqingyao Did you set the `RISCV_PATH` environment variable as described in the readme?

jim-wilson, 2019-05-21:
That looks like the right link.  What does 'riscv64-unknown-elf-gcc --print-multi-lib' say?

sunqingyao, 2019-05-21:
@nategraff-sifive Oh of course I ... Wait a second. I forgot to source the `.zshrc` so `RISCV_PATH` wasn't set in the current seesion!

Thanks for the heads up. It works now!

closed 2019-05-21 by nategraff-sifive

6357o31337, 2023-02-28:
@jim-wilson

Hello.

'riscv64-unknown-elf-gcc --print-multi-lib'
say

.;

Is this means that multilib is not enabled?
Then how to enable multilib?
I want to know how to configure using '--enable-multilib'.

I have almost similar with the fallowing.

Thank you.

> I'm having a similar issue on macOS Mojave.
> 
> Error message:
> 
> ```
> ~/Developer/freedom-e-sdk master#v201903?
> ❯ make software
> cd /Users/nalzok/Developer/freedom-e-sdk/bsp/sifive-hifive1/build/debug/ && \
> 		CFLAGS="-march=rv32imac -mabi=ilp32 -mcmodel=medlow -ffunction-sections -fdata-sections -I/Users/nalzok/Developer/freedom-e-sdk/bsp/sifive-hifive1/install/include -O0 -g" \
> 		/Users/nalzok/Developer/freedom-e-sdk/freedom-metal/configure \
> 		--host=riscv64-unknown-elf \
> 		--prefix=/Users/nalzok/Developer/freedom-e-sdk/bsp/sifive-hifive1/install \
> 		--libdir=/Users/nalzok/Developer/freedom-e-sdk/bsp/sifive-hifive1/install/lib/debug \
> 		--disable-maintainer-mode \
> 		--with-preconfigured \
> 		--with-machine-name=sifive-hifive1 \
> 		--with-machine-header=/Users/nalzok/Developer/freedom-e-sdk/bsp/sifive-hifive1/metal.h \
> 		--with-machine-ldscript=/Users/nalzok/Developer/freedom-e-sdk/bsp/sifive-hifive1/metal.default.lds \
> 		--with-builtin-libgloss
> checking for a BSD-compatible install... /usr/bin/install -c
> checking whether build environment is sane... yes
> checking for riscv64-unknown-elf-strip... riscv64-unknown-elf-strip
> checking for a thread-safe mkdir -p... /Users/nalzok/Developer/freedom-e-sdk/freedom-metal/install-sh -c -d
> checking for gawk... gawk
> checking whether make sets $(MAKE)... yes
> checking whether make supports nested variables... yes
> checking whether to enable maintainer-specific portions of Makefiles... no
> checking for riscv64-unknown-elf-gcc... riscv64-unknown-elf-gcc
> checking whether the C compiler works... no
> configure: error: in `/Users/nalzok/Developer/freedom-e-sdk/bsp/sifive-hifive1/build/debug':
> configure: error: C compiler cannot create executables
> See `config.log' for more details
> make: *** [/Users/nalzok/Developer/freedom-e-sdk/bsp/sifive-hifive1/build/debug/Makefile] Error 77
> ```
> 
> Log file: [config.log](https://github.com/sifive/freedom-e-sdk/files/3201024/config.log)
> 
> The toolchain (GNU Embedded Toolchain — v2019.02.0) is downloaded from https://www.sifive.com/boards
> 
> ```
> ~/Developer/freedom-e-sdk master#v201903?
> ❯ riscv64-unknown-elf-gcc --version
> riscv64-unknown-elf-gcc (GCC) 8.3.0
> Copyright (C) 2018 Free Software Foundation, Inc.
> This is free software; see the source for copying conditions.  There is NO
> warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
> ```

jim-wilson, 2023-02-28:
I'm "retired" for now, not checking email very often, and not actively working on the toolchain.  I haven't built freedom-e-sdk in a long time.  Yes, the ".;" output from --print-multi-lib means you have a toolchain without multilib support.  You can look for a pre-built one that includes multilib support.  Or try building one from source using --enable-multilib at configure time.  Building it from source is complicated, and may take some work if you have never done it before.  The toolchain is not part of freedom-e-sdk.

gcc 8.3.0 is an old toolchain.  I don't know if SiFive is still releasing GNU toolchains for public use.  You might need to look elsewhere.

For the freedom-e-sdk error, you should look at the config.log file for a clue as to what went wrong.  See the error message:
  configure: error: in `/Users/nalzok/Developer/freedom-e-sdk/bsp/sifive-hifive1/build/debug':
  configure: error: C compiler cannot create executables
  See `config.log' for more details
So look for the /Users/nalzok/Developer/freedom-e-sdk/bsp/sifive-hifive1/build/debug/config.log file and check for the error after it tests to see if the C compiler can create executables.  Configure does feature testing, and not all compilers have all features, so it is normal for there to be a bunch of optional tests some of which fail and some succeed, until it comes to a test failure that it can't work around.  So look for the last test failure and see why it failed.  It should be near the string "C compiler cannot create executables" in the config.log file.

6357o31337, 2023-03-01:
@jim-wilson 
Thank you a lot for answering even though you are retired.
No problem on late answer and so on.

'README.md' says 'The RISC-V GNU Toolchain available from the SiFive Website at https://www.sifive.com/software'
I can't found download file.

So, I did 'apt install gcc-riscv64-unknown-elf', and have the result of '--print-multi-lib' like below

[image]

Though still have similar result, but the 'config.log' changed a lot.

[image]

Any expert help me?
Thank you.

jim-wilson, 2023-03-02:
So looking at the config.log file I see

configure:2929: riscv64-unknown-elf-gcc -march=rv32imac -mabi=ilp32 -mcmodel=medlow -ffunction-sections -fdata-sections -I/home/x/freedom/e-sdk/freedom-e-sdk/bsp/freedom-e310-arty/install/include --specs=nano.specs -DMTIME_RATE_HZ_DEF=32768 -Os   conftest.c  >&5
riscv64-unknown-elf-gcc: fatal error: cannot read spec file 'nano.specs': No such file or directory
compilation terminated.

The freedom-e-sdk makefile wants to use --specs=nano.specs, but apparently that is missing or not supported by the toolchain that you are using.   The first thing to do is to try a basic sanity test.  Does the compiler work at all?  It looks like the Ubuntu riscv64-elf cross compiler does not work by default.

rohan:2088$ cat tmp.c
int main (void) { return 0; };
rohan:2089$ /usr/bin/riscv64-unknown-elf-gcc tmp.c
/usr/lib/riscv64-unknown-elf/bin/ld: cannot find crt0.o: No such file or directory
/usr/lib/riscv64-unknown-elf/bin/ld: cannot find -lc
/usr/lib/riscv64-unknown-elf/bin/ld: cannot find -lgloss
collect2: error: ld returned 1 exit status
rohan:2090$ 

But I see a picolibc-riscv64-unknown-elf package, and inside that package is a picolibc.specs file.  So after install that package, this works.

rohan:2090$ cat tmp.c
int main (void) { return 0; };
rohan:2091$ /usr/bin/riscv64-unknown-elf-gcc --specs=picolibc.specs tmp.c
rohan:2092$ file a.out
a.out: ELF 64-bit LSB executable, UCB RISC-V, RVC, double-float ABI, version 1 (SYSV), statically linked, with debug_info, not stripped
rohan:2093$ 

I know that this package is maintained by a Debian contributor.  Maybe Ubuntu doesn't have all of the packages imported from Debian.  Or maybe this is how the Debian contributor wants it to work.

Anyways, this doesn't help you build freedom-e-sdk.  freedom-e-sdk was developed using riscv-gnu-toolchain from the RISC-V International github tree.
    https://github.com/riscv-collab/riscv-gnu-toolchain
You can try building that from source.

As for SiFive, their web site has been reorganized a number of times, and it looks like the software page isn't useful anymore.  It only points at scs which doesn't appear to have any toolchain software, it only has cores.  You can still find toolchains on board specific pages.  So for instance
    https://www.sifive.com/boards/hifive1-rev-b
has toolchain links that still work and these toolchains have nano libc support.  I downloaded the Ubuntu one and it passed a trivial usage test so it appears to be still working.

6357o31337, 2023-03-03:
@jim-wilson 
Thank you a lot.

I solve my problem following what you mentioned as follows.

> As for SiFive, their web site has been reorganized a number of times, and it looks like the software page isn't useful anymore. It only points at scs which doesn't appear to have any toolchain [...]