sifive/freedom-e-sdk #230: Possible bug/implementation issue in freedom-metal

AmeyaVS, 2019-04-08:
Hello,

I am facing issue building the sample demo applications for SiFive HiFive1 board.
I have mentioned this issue in the riscv-gnu-toolchain repository reference can be found [here](https://github.com/riscv/riscv-gnu-toolchain/issues/423).

As per comments from @jim-wilson It seems the issue is with the usage of CLIC CSRs in the freedom-metal sources which has CLINT as per the specifications of the FE310 SoC.

In the meantime I will look for the riscv toolchain binary release which should be compatible with freedom-e-sdk build.

Regards,
Ameya Vikram Singh

palmer-dabbelt, 2019-04-08:
Thanks. This is a freedom-metal bug so I opened one over there.

knightsifive, 2020-11-11:
SiFive now provides prebuilt toolchains [here](https://www.sifive.com/software). If this issue persists when using a prebuilt toolchain, please let us know.

closed 2020-11-11 by knightsifive