sifive/freedom-e-sdk #236: SPI interfacing validation

girijabhagwat, 2019-04-16:
Hi, I am trying to implement SPI interface. I am referring the provided example /freedom-e-sdk/software/example-spi. I am trying to validate if my transfer is successfully completed or not. Does anyone facing any issues in transferring data? How should i validate? Is there any log file for that?

nategraff-sifive, 2019-04-16:
Hi @girijabhagwat,

What board are you using?

You might consider probing the SPI pins with a logic analyzer or oscilloscope.

girijabhagwat, 2019-04-16:
I am using Arty 100T board. I am debugging it with olimex cable. I am getting following error. I could upload program initially but I am getting following error now. I checked with oscilloscope then I was not getting any signal. What does that mean?

scripts/upload --elf /home/hardwaresec/Artix/freedom-e-sdk/software/hello/debug/hello.elf --openocd /home/hardwaresec/Artix/riscv-openocd-0.10.0-2018.12.0-x86_64-linux-ubuntu14/bin/openocd --gdb /home/hardwaresec/Artix/riscv64-unknown-elf-gcc-8.1.0-2019.01.0-x86_64-linux-ubuntu14/bin/riscv64-unknown-elf-gdb --openocd-config bsp/freedom-e310-arty/openocd.cfg
Open On-Chip Debugger 0.10.0+dev-gcd32bd4 (2018-12-19-01:15)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 10000 kHz
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 10000 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0x20000913 (mfg: 0x489 (SiFive, Inc.), part: 0x0000, ver: 0x2)
Info : datacount=1 progbufsize=16
Info : Disabling abstract command reads from CSRs.
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, misa=0x40001105
Info : Listening on port 3333 for gdb connections
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : accepting 'gdb' connection on tcp/3333
Error: timeout
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected
Remote connection closed
"monitor" command not supported by this target.
"monitor" command not supported by this target.
You can't do that when your target is `exec'
"monitor" command not supported by this target.
"monitor" command not supported by this target.
shutdown command invoked.

Is there any other way to debug it through command prompt?

nategraff-sifive, 2019-04-16:
What bitstream are you using on the Arty A7 100T? Last I heard there wasn't a Freedom E310 bitstream for the 100T yet, and it looks like you're using `TARGET=freedom-e310-arty`.

girijabhagwat, 2019-04-16:
I am using the mentioned target. I could generate the bitstream. I could also run the spi code initially but now I am getting above error. It stopped working suddenly. I don't think so there is any issue with my olimex cable as it is detecting my riscv processor. This error I am not able to solve and hence cannot debug my code.

erikdanie, 2019-04-16:
Hey, so it seems like you may be dealing with a mismatch of the target and board. In freedom currently, we don't have any configurations that build for the arty100t, only the 35t. So if you are using the e300 image built from freedom and programming it onto a 100t, there are likely some pin incompatibilities which means it won't work. We are looking to add support for this soon in the future, but in the meantime, you can try signing up on our website for an Arty100t e310 evaluation, or get an Arty35T and program that.

girijabhagwat, 2019-04-16:
I could flash program riscv processor on e310. I am referring following -
[https://www.digikey.com/eewiki/display/LOGIC/Digilent+Arty+A7+with+Xilinx+Artix-7+Implementing+SiFive+FE310+RISC-V#DigilentArtyA7withXilinxArtix-7ImplementingSiFiveFE310RISC-V-Comments](url).

I could upload program initially. Those were working fine for e310 evaluation. I could see the output on terminal. Now while uploading I am getting this error and I am not able to debug it.

girijabhagwat, 2019-04-16:
could flash program riscv processor on e310. I am referring following -
https://www.digikey.com/eewiki/display/LOGIC/Digilent+Arty+A7+with+Xilinx+Artix-7+Implementing+SiFive+FE310+RISC-V#DigilentArtyA7withXilinxArtix-7ImplementingSiFiveFE310RISC-V-Comments

I could upload program initially. Those were working fine for e310 evaluation. I could see the output on terminal. Now while uploading I am getting this error and I am not able to debug it.

closed 2019-07-10 by nategraff-sifive

Emil808, 2021-12-01:
Hi, Currently having the same problem.

Trying to run `example-spi` on the Arty A7 100T. I am using what is currently available, the Freedom-E310-Arty, and following this [guide](https://forum.digikey.com/t/digilent-arty-a7-with-xilinx-artix-7-implementing-sifive-fe310-risc-v/13311). The scala configuration shows that there is an available SPI1 and SPI2, and an I2C peripheral. But trying to use those, there is no activity on the pins.

ShousaPhedro, 2021-12-09:
Hi, 
@Emil808 how did you try to use the SPI1 and SPI2 peripheral?

I'm currently in the same situation, i followed the guide and tried to run the  `example-spi`. 
I want to use the SPI1 or SPI2 but I only find the SPI0 configured in the device tree and in the platform and metal files. Even so, i get a store access fault exception when trying to set the SPI Flash Interface Control Register to zero ,e.g., to programmable IO mode. ( at the `sifive_spi0.c` file, `METAL_SPI_REGW(METAL_SIFIVE_SPI0_FCTRL) = METAL_SPI_CONTROL_IO;`)

gette994, 2022-02-15:
Hi, 
@ShousaPhedro I'm currently also trying to get the ```example-spi``` to work. I have the same error as you, could you solve the problem in the meantime?

Emil808, 2022-02-15:
@ShousaPhedro, sorry for the delay, check the pull request I just made.

gette994, 2022-02-18:
@Emil808 thanks for the answer. I have added i2c, spi1 and spi2 in the core.dts file. Now I am able to run the ``example spi`` but I cannot measure anything on these pins on the A7 T100. 
I have also tried testing i2c. To run the ``example-i2c`` I used a different sensor, but that doesn`t work for me either (there is no signal on the i2c pins a4 and a5 either).
However, the pins can be controlled as normal gpios, I tested that. 
Do you know if there is any issue in the i2c or spi library?

Emil808, 2022-02-18:
@gette994, did you also made the appropriate modifications to the metal.h, metal-inline.h and metal-platform.h files? 

Is the `metal_spi_get_device(x)` or `metal_i2c_get_device(x)` returning the correct device handler, or are they returning null?

gette994, 2022-02-21:
@Emil808 yes, I changed the core.dts file and rebuilt the metal.h, metal-inline.h and metal-platform.h files with the make command make-bsp.

And yes the correct device handler is returned for both I2C and SPI.

Emil808, 2022-02-21:
@gette994
For the I2C, you need to modify these files from the [freedom repo](https://github.com/sifive/freedom). Then build implementation and re-program the FPGA. 

`src/main/scala/everywhere/e3e00artydevkit/FPGAChip.scala` You need to set up and assign the I2C pins to the desired board pins. Also set up internal pull-up resistors for the I2C lines, or you can set it up physically on your own. 

`fpga-shells/xilinx/arty_a7_100/constraints/arty-master.xdc` Un-comment the lines for the I2C lines. Make sure that the PACKAGE_PIN matches with the physical pin that you want to use on the FPGA board. 

Note that in `FPGAChip.scala`, the first argument of IOBUF() is the get_ports label in the `.xdc` file. The second argument is the pin label from the e310.

gette994, 2022-02-28:
@Emil808 thanks for the support I got it working now. 

However, I am still having problems using the Arduino IDE. At the moment it runs with Freedom Studio, but when I want to use the Arduino IDE, I get the following error message when uploading the program:

Open On-Chip Debugger 0.10.0-dev-g9bab078 (2017-02-02-01:39)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 10000 kHz
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 10000 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0x20000913 (mfg: 0x489 (<unknown>), part: 0x0000, ver: 0x2)
Error: Unsupported DTM version 1. (dtmcontrol=0x5071)
Error: Target not examined yet

Do you know how to fix the problem with the unsupported DTM version?