sifive/freedom-e-sdk #248: sifive-welcome source does not match the HiFive1 RevB docs

brucehoult, 2019-05-11:
The Getting Started document shows sifive-welcome printing "Are the LEDs Changing [y/n]" and responding to the user typing a "y".

This functionality is not present in the sifive-welcome source code. (it was in the old led_fade)

nategraff-sifive, 2019-05-13:
The program that ships with the HiFive1 Rev B is [this Zephyr RTOS app](https://github.com/sifive/hifive1-revb-welcome), so this is a HiFive1 Rev B docs bug.

brucehoult, 2019-05-13:
Who does the docs, is there a repo for it?

Can a Zephyr app conveniently read from the tty?

nategraff-sifive, 2019-05-13:
I'm not sure, since I haven't seen the document you're referencing. Maybe @JimSughrue can help?

Zephyr apps can read from the TTY, pretty sure, but that doesn't change the fact that the app that ships with the board won't have that functionality.