sifive/freedom-e-sdk #260: Hello world example does not work when run with QEMU

twilco, 2019-05-26:
Hi there!  Thanks for creating this great project.  I appreciate all the work you all are doing for the RISC-V ecosystem 🙂 .

---

Now to my issue...

Since I don't have a physical RISC-V core, I was attempting to test out the `hello` example from QEMU, ideally seeing the "Hello world!" pop up in my host terminal.  However, when I do the following:

```
git clone git@github.com:sifive/freedom-e-sdk.git
cd freedom-e-sdk
make software
qemu-system-riscv32 -nographic -machine sifive_e -kernel software/hello/debug/hello.elf
```

I get this error output:

```
$ qemu-system-riscv32 -nographic -machine sifive_e -kernel software/hello/debug/hello.elf
qemu: hardware error: sifive_prci_read: read: addr=0xc

CPU #0:
 pc       20403b64
 mhartid  00000000
 mstatus  00000000
 mip      00000000
 mie      00000000
 mideleg  00000000
 medeleg  00000000
 mtvec    20401af4
 mepc     00000000
 mcause   00000000
 zero 00000000 ra   2040371e sp   800013e0 gp   80001890
 tp   00000000 t0   20404c90 t1   00000000 t2   80001108
 s0   80001400 s1   00000001 a0   80001084 a1   0000000c
 a2   80000fd8 a3   20403b64 a4   10008000 a5   1000800c
 a6   0000001f a7   00000000 s2   20404c84 s3   00000000
 s4   00000000 s5   00000000 s6   00000000 s7   00000000
 s8   00000000 s9   00000000 s10  00000000 s11  00000000
 t3   00000000 t4   00000000 t5   00000000 t6   00000000
 ft0  0000000000000000 ft1  0000000000000000 ft2  0000000000000000 ft3  0000000000000000
 ft4  0000000000000000 ft5  0000000000000000 ft6  0000000000000000 ft7  0000000000000000
 fs0  0000000000000000 fs1  0000000000000000 fa0  0000000000000000 fa1  0000000000000000
 fa2  0000000000000000 fa3  0000000000000000 fa4  0000000000000000 fa5  0000000000000000
 fa6  0000000000000000 fa7  0000000000000000 fs2  0000000000000000 fs3  0000000000000000
 fs4  0000000000000000 fs5  0000000000000000 fs6  0000000000000000 fs7  0000000000000000
 fs8  0000000000000000 fs9  0000000000000000 fs10 0000000000000000 fs11 0000000000000000
 ft8  0000000000000000 ft9  0000000000000000 ft10 0000000000000000 ft11 0000000000000000
[1]    33111 abort      qemu-system-riscv32 -nographic -machine sifive_e -kernel
```

I'm using v8.2.0 of the GNU `riscv64-unknown-elf` toolchain, which I downloaded as a prebuilt toolchain from here: https://www.sifive.com/boards.  My OS is MacOS Mojave, v10.14.5.  My version of QEMU is:

```
$ qemu-system-riscv32 --version
QEMU emulator version 4.0.0
```

What am I doing incorrectly?

jim-wilson, 2019-05-26:
Unfortunately we don't have anyone actively working on qemu at the moment.  And we aren't actively using qemu for freedom-e-sdk work internally, though we hope to have this for a future release.  I believe this problem is a known bug that got fixed in the riscv.org qemu, was submitted upstream, was rejected for wrong code formatting, and has not yet been fixed and resubmitted.  I think this is the patch you need
http://lists.nongnu.org/archive/html/qemu-riscv/2019-03/msg00224.html
There may also be other problems.

nategraff-sifive, 2019-07-10:
You should be able to have some more success using [this branch of qemu](https://github.com/sifive/riscv-qemu/tree/riscv-qemu-3.1) and building for `TARGET=qemu-sifive-e31`.

You can also try out SiFive Freedom Studio which includes a prebuilt QEMU with the proper hardware support.

Closing the issue, feel free to reopen if you experience more problems.

closed 2019-07-10 by nategraff-sifive

Kanishk-K-U, 2022-11-27:
Hi, I created the same sample hello project in SiFive Freedom Studio with the target set to `qemu-sifive-e31` and the debug launch configuration set to "QEMU," but when I try to debug it, an error message stating that `QEMU failed with code (-1)`. appears. I couldn’t find any details on console.
<p>
[image]
</p>