sifive/freedom-e-sdk #275: Some doubts about "make upload".

clx7913, 2019-06-07:
Which address is the application(like gpio-demo) written at? 

If that address is the base address of flash, will it override the .mcs file which is downloaded into the flash before the setup of SoC for FPGA configuration?

nategraff-sifive, 2019-06-07:
Hi @clx7913,

For FPGA targets, the program should be loaded into the flash starting at the FPGA's reset vector, which shouldn't conflict with the stored FPGA configuration. For the Arty FPGA targets in Freedom E SDK, that should be at offset 0x400000 from the base of the SPI flash device.

You can see the address for you given target by looking at `bsp/<TARGET>/metal.default.lds`. You should see something like
```
MEMORY
{
	ram (wxa!ri) : ORIGIN = 0x80000000, LENGTH = 0x8000
	itim (wx!rai) : ORIGIN = 0x80008000, LENGTH = 0x8000
	flash (rxai!w) : ORIGIN = 0x40400000, LENGTH = 0x1fc00000
}
```
which for this target tells the linker to load the application at address `0x40400000`.

clx7913, 2019-06-10:
Thank you, sir.

I know the contents of the .lds files in this repository are rational, but could you take a look at this file [sirv-e-sdk/bsp/env/sirv-e201-arty/link.lds](url)?

I couldn't find an offset regard to the flash address (0x2000_0000) in the file above.

clx7913, 2019-06-10:
I still think there is not an expected offset in the file I quoted.

Maybe the openocd.cfg file will give me the answer.

closed 2019-07-10 by nategraff-sifive