sifive/freedom-e-sdk #282: How to transfer file on riscv-qemu from host?

lewiz-sunny, 2019-06-10:
hello I am following steps provided on 

https://www.cnx-software.com/2018/03/16/how-to-run-linux-on-risc-v-with-qemu-emulator/.

i was able to boot the qemu successfully, but now I want to run my own hello world program on qemu.

I have cross compiled my hello world program but, I am not able to find the directory to where i should place my executable file, so that I can access it from qemu after booting it up.

Any help is appreciated. thank you.

nategraff-sifive, 2019-06-10:
Hi @lewiz-sunny, unfortunately this isn't the right place to ask this question.

You might have better luck asking on the SiFive Forums though https://forums.sifive.com/.

Best wishes!

closed 2019-06-10 by nategraff-sifive

jim-wilson, 2019-06-10:
You have to copy the hello world program to the target file system.  On a linux host machine, you can do something like this:
sudo mount busybear.bin /mnt -o loop
sudo cp hello /mnt/root
sudo umount /mnt
then when you boot, you should find the program in /root, which is the root user home directory.  If you get network support working, then you can scp the file to the target.  This busybear info is old.  There are better alternatives now, but this should work for simple experiments.

lewiz-sunny, 2019-06-11:
@jim-wilson Thank you for your response. That really helped me.

lewiz-sunny, 2019-06-12:
@jim-wilson i am able to create a text file through my cross compiled user C code. 
Now I want to copy my text file from Qemu /root directory to my host system. Is there a way to do that? 
I did try copying the files but I am not getting the text inside my text file on the host system, whereas its working fine on the qemu. 

looking for your help. 

thank you

jim-wilson, 2019-06-12:
You should really start a new thread someplace else, as this isn't relevant to freedom-e-sdk.

If you can ssh into qemu, then you can use scp.  There is a slight option change, ssh -p becomes scp -P. 

If you don't have ssh, then you can use the same trick of mounting the qemu root file system via the loop device, as explained for busybear above.  Exactly how this works may depend on the format of your qemu root file system.

lewiz-sunny, 2019-06-12:
sure will start a thread somewhere else. I tried copying the file after mounting the image, but i am not able to copy the file under /root directory. @jim-wilson  thank you.