sifive/freedom-e-sdk #305: Writing code in assembly

zacharyselk, 2019-07-09:
Hi there, I was just wondering if there was any functionality for writing raw assembly (instead of c code) and using Freedom E SDK as a programmer & debugger. I tried poking around in the hello_world project makefile for a bit to see if I could get it to compile and run just an assembly file but it did not seem like it would be a simple thing to do.

Cheers,
Zach

nategraff-sifive, 2019-07-09:
Hi @zacharyselk,

You should be able to write raw assembly. Take a look at [the test-coreip example](https://github.com/sifive/test-coreip) (submoduled in this repository at `software/test-coreip`).

zacharyselk, 2019-07-09:
Fantastic, I will have to toy around with that. Thank you!

closed 2019-07-09 by zacharyselk