sifive/freedom-e-sdk #307: Insufficient BSP and not-working freedom-e310-arty

virtualboots, 2019-07-10:
Hi,

This repo's BSP contains very few targets (e.g., freedom-e310-arty, qemu-sifive-e31, ...). For Arty and RTLsim, although users can download them on the SiFive website and manually copy the bsp, it is very inconvenient to do so.

If the BSPs are public, could we consider just have a copy here.

Also, the freedom-e310-arty BSP in this repo is wrong. I'll have to use the one in SiFive E31 pre-built tar ball to make it work.

bsousi5, 2019-07-10:
Fu-Ching,

In regard to BSP files for standard cores Arty and RTL, what are you using for?
- browsing or references?
- run various examples and benchmark?

If you need them to be able to run examples and benchmark, it may more sense to create a script in freedom-e-sdk-infra to retrieve this from the saved location and add it to your workspace.

As for freedom-e310-arty BSP, can you update the BSP/dts file and create a PR.

Bunnaroath

virtualboots, 2019-07-10:
I'm viewing this from the customers' perspective. Wouldn't it be more convenient if customers can have all BSPs at once instead of downloading them one-by-one? Customers have no access to our private freedom-e-sdk-infra repo.

bsousi5, 2019-07-10:
From potential customer perspective, ie one wanting to evaluate our core, they typically know which core series they are interested.
They would either download one or two standard cores tar file, or sign up to SCD and design their own custom core.

Having said this, the primary reason we decided to not keep a copy standard BSP files in freedom-e-sdk is to ensure it is consistent with the RTL/FPGA bits streams available from sifive.com <http://sifive.com/>.

nategraff-sifive, 2019-07-10:
Closing to take this discussion internal.

closed 2019-07-10 by nategraff-sifive