sifive/freedom-e-sdk #328: Bootloader removed during restructure

jamolnng, 2019-08-06:
It appears that the safe bootloader for the hifive1 was removed during the restructure and move to metal. Could this be added back in the appropriate repository/location? While it can still be referenced through the commit history it should be added back to keep up with any new updates and for new boards. Maybe have each board's bootloader in their respective bsp folder?

nategraff-sifive, 2019-08-06:
Thanks for the suggestion, we'll work on it!

bradjc, 2020-04-30:
Is the bootloader code on the HiFive1 revb available anywhere?

jamolnng, 2020-06-12:
For the time being I have decided to attempt to reverse engineer the bootloader here https://github.com/jamolnng/hifive1b_bootloader

everything but the esp32 initialization is done

jamolnng, 2020-06-12:
I would not suggest overriding your current bootloader with my reverse engineer yet because it has not been sufficiently tested