sifive/freedom-e-sdk #389: Is there any available FreeRTOS  support on Arty

Aerotic, 2019-10-11:
According to some forum comments, I check out FreeRTOS branch but find nothing about FreeRTOS. And under V1_0FreeRTOS branch, I cannot pass the compilation.

bsousi5, 2019-10-11:
Hi Aero,

FreeRTOS has been a popular request and we are actively working to have it available officially. Please stay tune.

But in regard to your compilation failure in using the V1_0FreeRTOS branch, if you can describe the exact nature of the failure, someone may be able to response.

Regards,
Bunnaroath

Aerotic, 2019-10-12:
> Hi Aero, FreeRTOS has been a popular request and we are actively working to have it available officially. Please stay tune. But in regard to your compilation failure in using the V1_0FreeRTOS [...]

Hi Bunnaroath,

Thanks a lot, I'll later post the failure info here.

I find that in V1_0FreeRTOS branchthe, bsp/env includes freeedom-e300-arty. While in V201908-branch , there's a similiar bsp named freeedom-e310-arty. What's the difference of the two?
I'm using Arty-A7 with latest RocketChip RTL inside, can I upload software of freeedom-e300-arty to freeedom-e310-arty ?

By the way, what's release data of FreeRTOS for freeedom-e310-arty?

Regards,
Aero