sifive/freedom-e-sdk #424: Dhrystone benchmark target not working for freedom-e310-arty

ecroffleps, 2020-01-08:
I might be missing something obvious here, but i was trying to build the Dhrystone benchmark for the "freedom-e310-arty" target with following command:

`make PROGRAM=dhrystone TARGET=freedom-e310-arty software LINK_TARGET=ramrodata`

and got these errors during linking (: 

> /home/anon/riscv64/bin/riscv64-unknown-elf-gcc -march=rv32imac -mabi=ilp32 -mcmodel=medlow -ffunction-sections -fdata-sections -I/home/anon/freedom-sdk-new/bsp/freedom-e310-arty/install/include --specs=nano.specs -Os -O3 -DTIME -DNOENUM -Wno-implicit -save-temps -fno-builtin-printf -fno-common -falign-functions=4 -mexplicit-relocs -fno-inline -DDHRY_ITERS=20000000 -Xlinker --defsym=__stack_size=0x800 -Xlinker --defsym=__heap_size=0x1000 dhry_1.c dhry_2.c -Wl,--gc-sections -Wl,-Map,dhrystone.map -nostartfiles -nostdlib -L/home/anon/freedom-sdk-new/bsp/freedom-e310-arty/install/lib/release/ -T/home/anon/freedom-sdk-new/bsp/freedom-e310-arty/metal.ramrodata.lds  -Wl,--start-group -lc -lgcc -lm -lmetal -lmetal-gloss -Wl,--end-group -o dhrystone
dhry_1.c:35:14: warning: conflicting types for built-in function 'malloc' [-Wbuiltin-declaration-mismatch]
 extern char     *malloc ();
              ^~~~~~
dhry_1.c: In function 'main':
dhry_1.c:102:3: warning: incompatible implicit declaration of built-in function 'strcpy'
   strcpy (Ptr_Glob->variant.var_1.Str_Comp,
   ^~~~~~
dhry_1.c:102:3: note: include '<string.h>' or provide a declaration of 'strcpy'
dhry_1.c:68:1:
+#include <string.h>
 long            Begin_Time,
dhry_1.c:102:3:
   strcpy (Ptr_Glob->variant.var_1.Str_Comp,
   ^~~~~~
/home/anon/riscv64/bin/../lib/gcc/riscv64-unknown-elf/8.3.0/../../../../riscv64-unknown-elf/bin/ld: dhrystone section `.stack' will not fit in region `ram'
/home/anon/riscv64/bin/../lib/gcc/riscv64-unknown-elf/8.3.0/../../../../riscv64-unknown-elf/bin/ld: region `ram' overflowed by 4640 bytes

Im on the current master branch and using gcc version 8.3.0 (SiFive GCC 8.3.0-2019.08.0).

In the Readme the benchmark is build for "coreip-e31-arty" target but i can't find any reference to that BSP (here or on sifive.com).

nategraff-sifive, 2020-01-08:
Hi @ecroffleps,

There just isn't enough room on the `freedom-arty-e310` to fit Dhrystone with the current architecture of Freedom Metal. You can probably squeeze it in by carefully ditching a lot of the interrupt support, but unfortunately it's not too straightforward to find 4640 bytes and still have enough stack and heap for the benchmark to run and report its results properly.

As far as the README goes, `coreip-e31-arty` is a reference to an old target, you can disregard it.

ecroffleps, 2020-01-10:
Hi @nategraff-sifive,

thank you for the quick answer. 
That's unfortunate.

closed 2020-01-10 by ecroffleps