sifive/freedom-e-sdk #455: Question on example-pmp demo

yuhui-zheng, 2020-04-01:
I'm experimenting with the example-pmp demo https://github.com/sifive/example-pmp/blob/master/example-pmp.c. At line 100, it seems ```mcause``` register is not set for me and program doesn't enter ```store_access_fault_handler()```, which seems incorrect. I'm also observing ```protected_global[0]``` is not updated by executing line 100, which is the correct behavior. Could you confirm this is the intended behavior on SiFive HiFive1 revb? Thank you. 

I'm using 
```
SiFive Eclipse IDE for C/C++ Development
Version: 4.12.0.2019-08-2
```

taovcu, 2021-05-21:
I repeat the above tests on SiFive HiFive1 revb with same observations.

taovcu, 2021-05-21:
After I reset the board and run the original code again, I see the "store_access_fault_handler" is triggered. 
I add a printf("Entered store_access_fault_handler !\n") tag statement at the beginning of the handler function, I see the log is printed.

Virtual-Machine, 2021-06-15:
I also have an hifive1 revb and am having the same issue. In fact it wasn't mentioned above but after loading this program, I am forced to use safe mode to recover to a new program.  Just mentioning this in case someone else gets stuck after trying to run this.

To get into safe mode press 'reset' and when the green led flashes immediately press it again. If you time it right then it should start flashing red. This will now allow you to overwrite with a new program while led is flashing red.