sifive/freedom-e-sdk #567: make open doc happen error

sss22213, 2020-11-07:
when i make open doc, happen some error.

```
Opening /mnt/usb2/e_sdk/freedom-e-sdk/doc/html/index.html
gio: file:///mnt/usb2/e_sdk/freedom-e-sdk/doc/html/index.html: Error when getting information for file “/mnt/usb2/e_sdk/freedom-e-sdk/doc/html/index.html”: No such file or directory
make: *** [Makefile:173: open-docs] Error 4
```

i change directory into doc, and make. happen:

```
rm -rf html/* html/.buildinfo html/.doctrees
sphinx-build -b html sphinx html
Running Sphinx v1.8.5

Configuration error:
There is a programmable error in your configuration file:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/sphinx/config.py", line 368, in eval_config_file
    execfile_(filename, namespace)
  File "/usr/lib/python3/dist-packages/sphinx/util/pycompat.py", line 150, in execfile_
    exec_(code, _globals)
  File "/mnt/usb2/e_sdk/freedom-e-sdk/doc/sphinx/conf.py", line 27, in <module>
    version = os.environ['RELEASE_TAG']
  File "/usr/lib/python3.8/os.py", line 675, in __getitem__
    raise KeyError(key) from None
KeyError: 'RELEASE_TAG'

make: *** [Makefile:17: html/index.html] Error 2
```

megha-an, 2021-02-09:
There is a keyError for RELEASE_TAG variable 

because there is no environ variable with key name RELEASE_TAG

code:
 
 os.environ['RELEASE_TAG']  in /usr/lib/python3/dist-packages/sphinx/config.py is trying to fetch the RELEASE_TAG variable value , which is not present in os.environ
 
so it is throwing KeyError: 'RELEASE_TAG'

solution : 

I commented these two lines and run make 
it works

#version = os.environ['RELEASE_TAG']
#release = version

sss22213, 2021-02-10:
hi megha-an,
i found 
```python
version = os.environ['RELEASE_TAG']
```
on freedom-e-sdk/doc/sphinx/conf.py line 27

```python
release = version
```
on freedom-e-sdk/doc/sphinx/conf.py line 29

and than commit
it works, thanks.

closed 2021-02-10 by sss22213