sifive/freedom-e-sdk #570: printf with format 'lld' or 'llx' failed

zfchen819, 2020-11-20:
CPU: E21
toolchain: riscv64-unknown-elf-gcc-8.3.0-2020.04.0-x86_64-linux-ubuntu14.tar.gz
when I run hello.c, modify the code as below:
    **int main(){
        unsigned long long val = 0x123456789;
        printf("Hello, World! val:0x%llx\n",val);
    }**
but the run result is:
    **Hello, World! val:0xlx**
could someone help to solve this problem? thank you.

knightsifive, 2020-11-20:
SiFive's prebuilt toolchain provides two C libraries: Newlib and Newlib-Nano. To reduce code size, Newlib-Nano has reduced `printf` functionality. For example, it does not implement C99 formatting capabilities like the `ll` length modifier. If you use Newlib instead, it should have this capability. In previous versions of Freedom-E-SDK, this was as simple as deleting the `--specs=nano.specs` flag from `standalone.mk`. I don't know if this works in more recent versions however. And even if it does, the increased code size of Newlib (not -Nano) may cause linker errors.

zfchen819, 2020-11-20:
> SiFive's prebuilt toolchain provides two C libraries: Newlib and Newlib-Nano. To reduce code size, Newlib-Nano has reduced `printf` functionality. For example, it does not implement C99 formatting capabilities like the `ll` length modifier.
> 
> Freedom-E-SDK uses Newlib-Nano by default. If you switch to Newlib instead, you should gain this capability.
> 
> In previous versions of Freedom-E-SDK, switching to Newlib was as simple as deleting the `--specs=nano.specs` flag from `standalone.mk`. I don't know if this trick works in more recent versions, however. And even if it does, the increased code size of Newlib (not -Nano) may cause linker errors.

Thanks a lot for your perfect reply.
Now printf fuction works as expected. but another question want to ask you.

when I call getchar() function, it always return '-1',  i don't know what's the reason.

knightsifive, 2020-11-20:
I've never tried `getchar` before, sorry. I suggest you open a new issue, I will close this one.

closed 2020-11-20 by knightsifive