sifive/freedom-e-sdk #594: RISC-V toolchain - can't target rv32i & ilp32?

Slava0135, 2021-04-17:
[image]

nick-knight, 2021-04-17:
The RISC-V GNU toolchain certainly supports RV32I/ILP32. But the toolchain binaries that SiFive bundles with its "Freedom Tools" distribution (what you're using?) do not include "multilib" support for all possible ISA/ABI combinations. Instead, SiFive focuses on the combinations corresponding to its products, plus maybe a few others. If you have a use-case beyond the provided multilibs, you'll have to build your own toolchain. Head over to riscv/riscv-gnu-toolchain and learn how!

Slava0135, 2021-04-17:
I use version 10.2.0 of the pre-built toolchain.
Thanks for the response.

closed 2021-04-17 by Slava0135